prusa3d/Prusa-Firmware-ESP32-Cam #98: repeated watchdog resets after being on a while

AxelTong, 2024-12-24:
I've been having some issues with the ESP32-S3-CAM which are confusing me. 
It works perfectly when I first power it on, then after a while it doesn't upload to prusa connect anymore and the webpage also stops working. 
On checking the logs I found that the MCU is resetting due to watchdog repeatedly. Could you please help with this? 

Thanks. 

Here's the log (SSID has been manually removed, I'm willing to provide the full one privately): 
[SysLog.log](https://github.com/user-attachments/files/18240549/SysLog.log)

johnyHV, 2025-01-02:
Hello @AxelTong. Thanks for the log. Can you try for testing change photo resolution ? For example to 800x600 ?

AxelTong, 2025-01-03:
I don't have access for a few weeks, but I will once I get home. Thanks for looking into it already.