prusa3d/Prusa-Firmware-ESP32-Cam #99: Camera is working, but not sending image to prusa connect.

jaytonic, 2024-12-29:
Hi,

I did setup my camera(ESP32-S3-EYE), using the latest firmware(1.1.1).
It is properly connected to my wifi, and I've properly set the prusaconnect token.

The esp32 is near the wifi router, I'm able to start stream from the computer without any issue, but not image get sent to the prusa connect.

I checked the logs, and I get some error:

```
0000-00-00_00-00-00 - Taking photo...
0000-00-00_00-00-00 - The picture has been saved. Size: 64964 bytes, Photo resolution: 1600 x 1200
0000-00-00_00-00-00 - Photo OK! 0
0000-00-00_00-00-00 - Exif header OK! Len: 310
0000-00-00_00-00-00 - Start sending photo to prusaconnect
0000-00-00_00-00-00 - Sending Photo to PrusaConnect, 65254 bytes
0000-00-00_00-00-00 - Free RAM: 187808 B, Min: 173448
0000-00-00_00-00-00 - Free PSRAM: 7954668 B, Min: 7908216
0000-00-00_00-00-00 - MCU Temperature: 48.50 *C
0000-00-00_00-00-00 - Sync NTP time fail
0000-00-00_00-00-00 - WiFi status: Wifi connected. SSID: Skynet, BSSID: 9A:A6:7E:61:71:14, RSSI: -63 dBm, IP: 192.168.0.105, TX power: 20dBm
0000-00-00_00-00-00 - CardStatus: 1 FileStatus: 1
0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
0000-00-00_00-00-00 - Start sending device information to prusaconnect
0000-00-00_00-00-00 - Data: {"config":{"name":"prusa-esp32cam","firmware":"1.1.1","manufacturer":"Prusa","model":"ESP32-S3-EYE_2.2","resolution":{"width":1600,"height":1200},"network_info":{"wifi_mac":"8C:BF:EA:03:5B:D8","wifi_ipv4":"192.168.0.105","wifi_ssid":"Skynet"}}}
0000-00-00_00-00-00 - Sending Info to PrusaConnect, 244 bytes
```

First thing, it seems that the device is unable to get any NTP sync. 
*after some time, it does get an NTP sync and then it manages to send the image to prusa connect*
But it can wait for a very long time, or sometimes it doesn't work anymore after it worked for a moment.

Any idea what could be going on?

mozgy, 2025-01-02:
Check your DNS resolving config in WiFi router.

jaytonic, 2025-01-02:
> Check your DNS resolving config in WiFi router.

What do you mean? The DNS is coming from my ISP and the router doesn't allow to change it. But After some time(like 10-15min) it gets a an NTP sync. It's on the dame wifi than 4 computers, 4 phone and like 30 docker containers which all manages to get proper DNS. So I'm not really sure the router is to blame here? Can I manually set some DNS on the esp cam?

johnyHV, 2025-01-02:
Hello @jaytonic . Can you send me full log file from camera ?

mozgy, 2025-01-06:
> What do you mean? The DNS is coming from my ISP and the router doesn't allow to change it. But After some time(like 10-15min) it gets a an NTP sync. It's on the dame wifi than 4 computers, 4 phone [...]

DNS resolve and NTP sync should be in seconds if ISP properly propagates that info from BRAS (or similar) to your router. Not being able to sync NTP can be result of several reasons - DNS not resolving, NTP config wrong servers, router not set up for NTP cache/proxy, router firewall active. If you get NTP sync with big delay thats a telltale that you get response only from a second config entry. You can change NTP server config in 'mcu_cfg.h' <br>
/* ----------------- NTP CFG --------------------*/
#define NTP_SERVER_1                "pool.ntp.org"          ///< NTP server
#define NTP_SERVER_2                "time.nist.gov"         ///< NTP server
<br> but you have to recompile the whole fw.

jaytonic, 2025-01-11:
> Hello @jaytonic . Can you send me full log file from camera ?

Here. In this case, after a few minutes, I'm still not getting any image on the Prusa connect dashboard, while still being able to display the live stream directly on the camera. I guess that in 30-40min, I will have it.

> ----------------------------------------------------------------
> Start MCU!
> SW Version: 1.1.1 ,Build: Nov 25 2024 23:13:52
> Verbose mode: false
> Log level: 2
> 0000-00-00_00-00-00 - Init system lib
> 0000-00-00_00-00-00 - SW Version: 1.1.1 Build: Nov 25 2024 23:14:00
> 0000-00-00_00-00-00 - Board name: ESP32-S3-EYE_2.2
> 0000-00-00_00-00-00 - CPU reset reason: Reset due to power-on event
> 0000-00-00_00-00-00 - CPU0 reset reason: POWERON_RESET
> 0000-00-00_00-00-00 - CPU1 reset reason: POWERON_RESET
> 0000-00-00_00-00-00 - MCU Temperature: 18.50 *C
> 0000-00-00_00-00-00 - Internal Total heap: 351420 B, Internal Free Heap: 311320
> 0000-00-00_00-00-00 - PSRAM Total heap: 8388608 B, PSRAM Free Heap: 8358244
> 0000-00-00_00-00-00 - Chip model: ESP32-S3, ChipRevision: 2, Cpu Freq: 240
> 0000-00-00_00-00-00 - SDK Version: v5.1.4-972-g632e0c2a9f-dirty, Core Version: 3.0.7
> 0000-00-00_00-00-00 - Flash Size: 8388608, Flash Speed 80000000, Flash Mode: 2
> 0000-00-00_00-00-00 - PSRAM is used.
> 0000-00-00_00-00-00 - malloc/new is not using SPIRAM
> 0000-00-00_00-00-00 - Init cfg module: 393
> 0000-00-00_00-00-00 - Read FirstMcuStart FLAG
> 0000-00-00_00-00-00 - It's not first start MCU: 15
> 0000-00-00_00-00-00 - web auth enable: 0
> 0000-00-00_00-00-00 - web auth user: admin
> 0000-00-00_00-00-00 - web auth pass: ******
> 0000-00-00_00-00-00 - Load camera CFG from EEPROM
> 0000-00-00_00-00-00 - Photo quality: 10
> 0000-00-00_00-00-00 - Framesize: 6
> 0000-00-00_00-00-00 - Framesize: 6
> 0000-00-00_00-00-00 - brightness: 2
> 0000-00-00_00-00-00 - contrast: 0
> 0000-00-00_00-00-00 - saturation: 0
> 0000-00-00_00-00-00 - awb: 1
> 0000-00-00_00-00-00 - awb_gain: 1
> 0000-00-00_00-00-00 - awb_mode: 0
> 0000-00-00_00-00-00 - aec2: 1
> 0000-00-00_00-00-00 - ae_level: 0
> 0000-00-00_00-00-00 - aec_value: 300
> 0000-00-00_00-00-00 - gain_ctrl: 1
> 0000-00-00_00-00-00 - agc_gain: 0
> 0000-00-00_00-00-00 - bpc: 1
> 0000-00-00_00-00-00 - wpc: 1
> 0000-00-00_00-00-00 - raw_gama: 1
> 0000-00-00_00-00-00 - hmirror: 1
> 0000-00-00_00-00-00 - vflip: 1
> 0000-00-00_00-00-00 - lensc: 1
> 0000-00-00_00-00-00 - exposure_ctrl: 1
> 0000-00-00_00-00-00 - Camera flash: 0
> 0000-00-00_00-00-00 - Camera flash time: 200
> 0000-00-00_00-00-00 - Camera image rotation: 1
> 0000-00-00_00-00-00 - Load PrusaConnect CFG from EEPROM
> 0000-00-00_00-00-00 - Token: *********************
> 0000-00-00_00-00-00 - Fingerprint: MTQwMTkxMjM0MzkxMjE2IDAwOjAwOjAwOjAwOjAwOjAw
> 0000-00-00_00-00-00 - Refresh interval: 10
> 0000-00-00_00-00-00 - PrusaConnect hostname: connect.prusa3d.com
> 0000-00-00_00-00-00 - Time lapse function status: 0
> 0000-00-00_00-00-00 - SSID: Skynet
> 0000-00-00_00-00-00 - WiFi password: ************
> 0000-00-00_00-00-00 - mDNS: prusa-esp32cam
> 0000-00-00_00-00-00 - Enable Service AP: 1
> 0000-00-00_00-00-00 - Network IP method: DHCP
> 0000-00-00_00-00-00 - Network static IP: 255.255.255.255
> 0000-00-00_00-00-00 - Network static mask: 255.255.255.255
> 0000-00-00_00-00-00 - Network static gateway: 255.255.255.255
> 0000-00-00_00-00-00 - Network static DNS: 255.255.255.255
> 0000-00-00_00-00-00 - Service AP SSID: ESP32_camera_3.91.216
[... 445 lines cut ...]
> 0000-00-00_00-00-00 - Sync NTP time fail
> 0000-00-00_00-00-00 - WiFi status: Wifi connected. SSID: Skynet, BSSID: 9A:A6:7E:61:7A:2C, RSSI: -60 dBm, IP: 192.168.0.105, TX power: 20dBm
> 0000-00-00_00-00-00 - CardStatus: 1 FileStatus: 1
> 0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
> 0000-00-00_00-00-00 - Start sending device information to prusaconnect
> 0000-00-00_00-00-00 - Data: {"config":{"name":"prusa-esp32cam","firmware":"1.1.1","manufacturer":"Prusa","model":"ESP32-S3-EYE_2.2","resolution":{"width":1600,"height":1200},"network_info":{"wifi_mac":"8C:BF:EA:03:5B:D8","wifi_ipv4":"192.168.0.105","wifi_ssid":"Skynet"}}}
> 0000-00-00_00-00-00 - Sending Info to PrusaConnect, 244 bytes
> 0000-00-00_00-00-00 - Stream, average data in 30sec. FPS: 2.1, Size: 195KB
> 0000-00-00_00-00-00 - Free RAM: 178624 B, Min: 156288
> 0000-00-00_00-00-00 - Free PSRAM: 7735660 B, Min: 7702668
> 0000-00-00_00-00-00 - MCU Temperature: 52.50 *C
> 0000-00-00_00-00-00 - Waiting for NTP time sync: 
> 0000-00-00_00-00-00 - System time: 0000-00-00_00-00-00
> 0000-00-00_00-00-00 - Sync NTP time fail
> 0000-00-00_00-00-00 - WiFi status: Wifi connected. SSID: Skynet, BSSID: 9A:A6:7E:61:7A:2C, RSSI: -55 dBm, IP: 192.168.0.105, TX power: 20dBm
> 0000-00-00_00-00-00 - CardStatus: 1 FileStatus: 1
> 0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
> 0000-00-00_00-00-00 - Start sending photo to prusaconnect
> 0000-00-00_00-00-00 - Sending Photo to PrusaConnect, 199090 bytes
> 0000-00-00_00-00-00 - Stream, average data in 30sec. FPS: 2.0, Size: 196KB
> 0000-00-00_00-00-00 - Free RAM: 178212 B, Min: 156288
> 0000-00-00_00-00-00 - Free PSRAM: 7755292 B, Min: 7702668
> 0000-00-00_00-00-00 - MCU Temperature: 52.50 *C
> 0000-00-00_00-00-00 - Waiting for NTP time sync: 
> 0000-00-00_00-00-00 - System time: 0000-00-00_00-00-00

jaytonic, 2025-01-11:
And here when it did get an NTP sync and started working:

> 0000-00-00_00-00-00 - System time: 0000-00-00_00-00-00
> 0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
> 0000-00-00_00-00-00 - Start sending device information to prusaconnect
> 0000-00-00_00-00-00 - Data: {"config":{"name":"prusa-esp32cam","firmware":"1.1.1","manufacturer":"Prusa","model":"ESP32-S3-EYE_2.2","resolution":{"width":1600,"height":1200},"network_info":{"wifi_mac":"8C:BF:EA:03:5B:D8","wifi_ipv4":"192.168.0.105","wifi_ssid":"Skynet"}}}
> 0000-00-00_00-00-00 - Sending Info to PrusaConnect, 244 bytes
> 0000-00-00_00-00-00 - Sync NTP time fail
> 0000-00-00_00-00-00 - WiFi status: Wifi connected. SSID: Skynet, BSSID: 9A:A6:7E:61:7A:2C, RSSI: -57 dBm, IP: 192.168.0.105, TX power: 20dBm
> 0000-00-00_00-00-00 - CardStatus: 1 FileStatus: 1
> 0000-00-00_00-00-00 - Free RAM: 181100 B, Min: 156288
> 0000-00-00_00-00-00 - Free PSRAM: 7758304 B, Min: 7702668
> 0000-00-00_00-00-00 - MCU Temperature: 67.50 *C
> 0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
> 0000-00-00_00-00-00 - Taking photo...
> 0000-00-00_00-00-00 - The picture has been saved. Size: 208938 bytes, Photo resolution: 1600 x 1200
> 0000-00-00_00-00-00 - Photo OK! 0
> 0000-00-00_00-00-00 - Exif header OK! Len: 310
> 0000-00-00_00-00-00 - Start sending photo to prusaconnect
> 0000-00-00_00-00-00 - Sending Photo to PrusaConnect, 209228 bytes
> 0000-00-00_00-00-00 - Disable service AP mode
> 0000-00-00_00-00-00 - WiFi AP stop
> 0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 8
> 0000-00-00_00-00-00 - Connecting to STA BSSID
> 0000-00-00_00-00-00 - WiFi AP start
> 0000-00-00_00-00-00 - WiFi AP stop
> 0000-00-00_00-00-00 - WiFi AP start
> 0000-00-00_00-00-00 - WiFi AP stop
> 0000-00-00_00-00-00 - Reconnecting to WiFi. STA
> 0000-00-00_00-00-00 - Disconnect from WiFi
> 0000-00-00_00-00-00 - Reconnecting to WiFi. STA
> 0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
> 0000-00-00_00-00-00 - WiFi disconnected from access point. Reason: 201
> 0000-00-00_00-00-00 - WiFi connected to STA
> 0000-00-00_00-00-00 - WiFi Got IP address: 192.168.0.105
> 0000-00-00_00-00-00 - WiFi Got mask: 255.255.255.0
> 0000-00-00_00-00-00 - WiFi Got gateway: 192.168.0.1
> 0000-00-00_00-00-00 - WiFi Got DNS 1: 192.168.0.1
> 0000-00-00_00-00-00 - WiFi Got DNS 2: 0.0.0.0
> 0000-00-00_00-00-00 - Waiting for NTP time sync: 
> 0000-00-00_00-00-00 - System time: 0000-00-00_00-00-00
> 0000-00-00_00-00-00 - Start sending device information to prusaconnect
> 0000-00-00_00-00-00 - Data: {"config":{"name":"prusa-esp32cam","firmware":"1.1.1","manufacturer":"Prusa","model":"ESP32-S3-EYE_2.2","resolution":{"width":1600,"height":1200},"network_info":{"wifi_mac":"8C:BF:EA:03:5B:D8","wifi_ipv4":"192.168.0.105","wifi_ssid":"Skynet"}}}
> 0000-00-00_00-00-00 - Sending Info to PrusaConnect, 244 bytes
> 0000-00-00_00-00-00 - Send done: 244/244 bytes
> 0000-00-00_00-00-00 - Upload done. Response code: Info: 200 - OK ,BA:Backend available
> 0000-00-00_00-00-00 - Taking photo...
> 0000-00-00_00-00-00 - The picture has been saved. Size: 202209 bytes, Photo resolution: 1600 x 1200
> 0000-00-00_00-00-00 - Photo OK! 0
> 0000-00-00_00-00-00 - Exif header OK! Len: 310
> 0000-00-00_00-00-00 - Start sending photo to prusaconnect
> 0000-00-00_00-00-00 - Sending Photo to PrusaConnect, 202499 bytes
> 0000-00-00_00-00-00 - Send done: 202499/202499 bytes
> 0000-00-00_00-00-00 - Sync NTP time fail
> 0000-00-00_00-00-00 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
> 0000-00-00_00-00-00 - CardStatus: 1 FileStatus: 1
> 0000-00-00_00-00-00 - Taking photo...
> 0000-00-00_00-00-00 - The picture has been saved. Size: 172285 bytes, Photo resolution: 1600 x 1200
> 0000-00-00_00-00-00 - Photo OK! 0
> 0000-00-00_00-00-00 - Exif header OK! Len: 310
> 0000-00-00_00-00-00 - Start sending photo to prusaconnect
> 0000-00-00_00-00-00 - Sending Photo to PrusaConnect, 172575 bytes
> 0000-00-00_00-00-00 - Send done: 172575/172575 bytes
> 0000-00-00_00-00-00 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
> 0000-00-00_00-00-00 - Free RAM: 188984 B, Min: 135092
> 0000-00-00_00-00-00 - Free PSRAM: 7763452 B, Min: 7702668
> 0000-00-00_00-00-00 - MCU Temperature: 68.50 *C
> 0000-00-00_00-00-00 - Waiting for NTP time sync: 
> 0000-00-00_00-00-00 - System time: 0000-00-00_00-00-00
> 2025-01-11_11-10-32 - System time: 2025-01-11_11-10-32
> 2025-01-11_11-10-32 - Sync NTP time done. Set UTC timezone
> 2025-01-11_11-10-32 - WiFi status: Wifi connected. SSID: Skynet, BSSID: 9A:A6:7E:61:7A:2C, RSSI: -59 dBm, IP: 192.168.0.105, TX power: 20dBm
> 2025-01-11_11-10-36 - Taking photo...
> 2025-01-11_11-10-36 - The picture has been saved. Size: 188665 bytes, Photo resolution: 1600 x 1200
> 2025-01-11_11-10-36 - Photo OK! 0
> 2025-01-11_11-10-36 - Exif header OK! Len: 310
> 2025-01-11_11-10-36 - Start sending photo to prusaconnect
> 2025-01-11_11-10-36 - Sending Photo to PrusaConnect, 188955 bytes
> 2025-01-11_11-10-38 - Send done: 188955/188955 bytes
> 2025-01-11_11-10-39 - Upload done. Response code: Photo: 200 - OK ,BA:Backend available
> 2025-01-11_11-10-45 - CardStatus: 1 FileStatus: 1

hreality, 2025-02-15:
I had a similar issue and resolved it by looking under advanced settings for wifi.  I set the default gateway to my router IP and the deault DNS to 8.8.8.8 (google).  the defaults were all set to 255.255.255.255

genericprinter0815, 2025-04-27:
I had the same issue: Even after waiting for 15+ minutes, no successful NTP time sync happened.
Hardware: Freenove ESP32-S3-Wroom
FW version 1.1.2
@hreality led me to play with the Advanced Wifi settings (thanks for the hint!). The following worked for me:
Setting the **Wi-Fi client IPv4 Method** to **Manual (static IP)**
- IP address to 192.168.0.50 (or whatever you like to reach it on later)
- Subnet mask: 255.255.255.0 (check your router's IP Subnet Mask)
- Default gateway: Your router's Gateway IP address, e.g. 192.168.1.1, 192.168.0.1. Commonly 192.168.X.X
- DNS server: 1.1.1.1 (Cloudflare DNS. Google's 8.8.8.8 or others might work too - not tested).

Common log output messages before solving the issue:

```
0000-00-00_00-00-00 - Waiting for NTP time sync: 
0000-00-00_00-00-00 - System time: 0000-00-00_00-00-00
0000-00-00_00-00-00 - Sync NTP time fail
0000-00-00_00-00-00 - Failed connect to OTA server!
0000-00-00_00-00-00 - Connetion failed to domain! Error: -1 - ERROR - Generic error : 0 ,BA:Wait for first connection
```

jaytonic, 2025-09-05:
No, honestly, this camera setup is really garbage, never seen something that works so badly. Sometimes it works after 15-20 minutes of being on, and usually it continues working only it get turned off... Disappointed from prusa on this one.