prusa3d/Prusa-Firmware-ESP32-Cam #101: Add support for OV3660 cameras

tkerby, 2025-01-07:
Many of the newer ESP cameras use the ESP32S3 and the OV3660 camera (e.g. the M5Stack M12 Camera). This should be possible to support with some minor changes to the code.

dabijja, 2025-02-05:
Hello,

Any news on this? The latest vaersion of Freenove_ESP32_WROVER_Board is equiped with OV3660 and SD card.

Thank you

E-B3rry, 2025-03-15:
Hi, I'm currently using a Freenove ESP32-S3-Wroom with a OV3660 (couldn't find any data regarding the camera module where I ordered it, but it's written on the ribbon cable) and it worked straight out of the box. Are you sure it's not already supported for the WROVER too? It has [been tested and it apparently works](https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/blob/master/doc/ESP32-Wrover-dev/README.md#supported-camera-modules).

stephan14167, 2025-04-18:
Hello,
I have a Freenove ESP32-S3-Wroom with cam type OV3660 directly from expressif.
With the prusa software I only get a black image.
OV3660 with other software works.
Does anyone have any ideas?

15:59:46.232 -> 0000-00-00_00-00-00 - Init camera lib
15:59:46.232 -> 0000-00-00_00-00-00 - Init GPIO
15:59:46.232 -> E (4356) rmt: rmt_new_tx_channel (269) : not able to power down in light sleep
15:59:46.271 -> 0000-00-00_00-00-00 - Init camera module
15:59:46.432 -> 0000-00-00_00-00-00 - WiFi connected to STA
15:59:46.601 -> 0000-00-00_00-00-00 - Set camera CFG
15:59:46.635 -> 0000-00-00 00-00-00 - Get camera model and type
15:59:46.635 -> 0000-00-00 00-00-00 - Camera type: 13920
15:59:46.635 -> 0000-00-00 00-00-00 - Camera name: OV3660
15:59:46.635 -> 0000-00-00 00-00-00 - Camera model: 2
15:59:46.635 -> 0000-00-00 00-00-00 - Camera PID: 13920
15:59:46.684 -> 0000-00-00_00-00-00 - Camera MAX framesize: 19
15:59:46.684 -> 0000-00-00_00-00-00 - Camera support jpeg: 1
15:59:46.767 -> 0000-00-00_00-00-00 - Taking photo ...
15:59:46.835 -> 0000-00-00_00-00-00 - The picture has been saved. Size: 2178 bytes, Photo resolution: 320 x 240
15:59:46.835 -> 0000-00-00_00-00-00 - Photo OK! 0
15:59:46.835 -> 0000-00-00_00-00-00 - Exif header OK! Len: 310

tksly92, 2025-04-30:
I have the same problem.  Only see black image with the OV3660.

Dianoga, 2025-06-27:
I thought I was getting a solid black image but it turns out it was just very dark. When I turn the brightness up to full I can see things in the image but it's still too dark to be useful.

yasars, 2025-07-01:
OV3660 working good pls add 2048 x 1536 resolution...

rey17pro, 2025-07-19:
DO ANYONE KNOW HOW TO USE THIS FOR STREAMING VIDEO FROM ESP32 S3 WROOM 1 WITH OV3660 RIBBON CAMERA ATTACHED