prusa3d/Prusa-Firmware-ESP32-Cam #104: frequent watchdog and panic restarts on ESP32_S3_WROOM_FREENOVE

boschhd, 2025-02-01:
I've followed the documentation on Freenove ESP32-S3-Wroom and there were no obvious errors in the process.

Tried installing via supplied binaries or compiling with arduino IDE. With the same effect:
The web UI is unusably slow and times out often (it's hard to download logs) and the serial monitor shows frequent reboots due to Watchdog (CaptureAndSendP) or exceptions/panic (Backtrace: 0xfffffffe:0x80380d90 |<-CORRUPTED or Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 

I attached some snippets from the log (Arduino IDE allows to copy only one screen at a time), and a downloaded log for context.

Maybe my hardware is just faulty?

Thanks for any hint and best regards,
Harald

[log2.txt](https://github.com/user-attachments/files/18626795/log2.txt)
[log3.txt](https://github.com/user-attachments/files/18626796/log3.txt)
[log4.txt](https://github.com/user-attachments/files/18626798/log4.txt)
[log1.txt](https://github.com/user-attachments/files/18626797/log1.txt)

[get_logs.txt](https://github.com/user-attachments/files/18626800/get_logs.txt)

boschhd, 2025-02-01:
I think I found the culprit.
The Freenove ESP32S3 has an interference problem between wifi and camera. It seems that, if the camera is used, there is a high frequency on some pins which kills the onboard wifi antenna, leading to the above mentioned errors and timeouts.

One can easily test for this: if you try to stream video and it fails, but when you touch the pins around the 3V3 pin, then it suddenly works.

Workaround: set the xclk frequency to 8 MHz, then the interference goes away.
I changed the line 110 in camera.cpp to 8000000 and now everything runs smoothly :)

Maybe you can include this info into the doc.
It seems to affect other boards as well

Sources: 
Touch the correct part of the board: https://www.reddit.com/r/esp32/comments/c4iy1j/aithinker_esp32_cam_slow_video_stream_try/ 
High frequency issue: https://www.reddit.com/r/esp32/comments/r9g5jc/fixing_ymmv_the_poor_frame_rate_on_the_esp32cam/
Indication that it might concern channel 1: https://github.com/espressif/esp-idf/issues/14008
8Mhz workaround: https://github.com/espressif/arduino-esp32/issues/4655#issuecomment-964595203

krystof-k, 2025-03-09:
I was having the same issues with Seeed Studio XIAO ESP32S3 Sense – WiFi not working until touched the antenna or disconnected the camera. But suddenly, when I wanted to record a video, it started working just fine. I don't get it.

E-B3rry, 2025-03-15:
Can confirm that it works, solved the issue with Freenove ESP32-S3-WROOM. Thank you!

genericprinter0815, 2025-04-27:
Solved my issue of a very unstable camera web server too!

staze, 2025-08-07:
omg, thank you for this. board even running cooler than it was. before was also getting ssl errors.

mimeister, 2025-08-16:
Thanks a lot. I'm using the ESP32-S3-CAM and it constantly kept running into watchdog resets. Changing my WiFi's channel (was channel 1 before) did not help, but compiling myself with the suggested change to the cam's xclk worked great.
The cam doesn't crash anymore and the web UI is much more responsive. Before, it was very slow and any interaction could cause another crash.

jonisu, 2025-10-19:
Thanks for the suggestion to set the xclk frequency to 8 MHz! My camera works like a charm now.