prusa3d/Prusa-Firmware-ESP32-Cam #112: Boot loop - jpeg not a supported format?

pjvenda, 2025-03-17:
Hello all,
I tried the firmware today on an esp32 s3 Dev device with an ov2640 camera (or so the doc said). I compiled it from the source on Arduino ide and it compiled and uploaded correctly.

But it is in a boot loop, camera initialisation fails (error 106) because jpeg is not a supported format (?).

I have another board arriving today so I should be able to swap the camera and try again. Failing that, what should I try next?

(Some logs to follow)

I installed all libraries on the Arduino ide interface which is not exactly as per the instructions but the firmware compiled ok and the error message suggests an issue with the camera...

pjvenda, 2025-03-17:
```
(...)
0000-00-00_00-00-00 - 10 | EE WiFi                          |  -81 |  1 | REDACTED
0000-00-00_00-00-00 - 11 | BT-5JCPKM                        |  -82 |  1 | REDACTED
0000-00-00_00-00-00 - SSID:  not found
0000-00-00_00-00-00 - Starting mDNS record: http://prusa-esp32cam.local
0000-00-00_00-00-00 - Starting mDNS OK
0000-00-00_00-00-00 - Init camera lib
0000-00-00_00-00-00 - Init GPIO
0000-00-00_00-00-00 - Init camera module
E (3976) camera: JPEG format is not supported on this sensor
E (3976) gdma: gdma_disconnect(314): no peripheral is connected to the channel
0000-00-00_00-00-00 - Camera init failed. Error: 106
0000-00-00_00-00-00 - Reset ESP32-cam!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047687
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
----------------------------------------------------------------
Start MCU!
```

pjvenda, 2025-03-17:
I can now confirm this is a false positive. The hardware I had was not an OV2640 camera. Apologies for the noise.

closed 2025-03-17 by pjvenda

sandeepsonskar, 2026-10-05:
try this 
https://github.com/sandeepsonskar/esp32-RHYX-M21-45-CAM-.git